MoonshineSG/OctoPrint-MultiColors #3: Example of gcode to pause and resume

guysoft, 2016-07-12:
Hey,
I am trying to understand how to use this plugin.

Would something like this work?
Should I change it in any way?

```
M83 ; turn on relative movement for extruder  
G1 E-5.000000 F6000 ; retract filament 5mm  
G1 X0.000000 Y0.000000 Z4.000000 F9000 ; home X and Y axis leave Z at current height  
M84 E ; release extruder stepper motor from 'holding' position  
@pause ; pause print!
G1 X0.000000 Y0.000000 F9000 ; upon resume, rehome X/Y in case position was bumped out  
G1 X69.080000 Y58.560000 F9000 ; move back to next layer starting position  
G1 E0 F6000 ; reset extruder, ready to push out plastic again  
G1 F9000  
M82 ; set extruder movement back to absolute ready for next layer
```

MoonshineSG, 2016-07-12:
Probably not... 

1. "Z4" will not " leave Z at current height" 
2. I don't know if your printer reacts to "@pause" 
3. there are many hardcoded values, so the gcode will not be generic. Might work if you change it for every print you use it for.. 

I simply insert an `M0` which will pause the print, move the hear to 0, 0 `G1 X0 Y0` (so that I can easily access the nozzle), purge some fillament `G1 E20 F300`  and then press `Resume` in Nautilus (OctoPrint on iPhone) . The next lines of gcode will move the head back to the old location, so I don;t need to care about that...

guysoft, 2016-07-12:
Can't the plugin itself pause he printer when reaching a certain layer?
My printer is a Pursa i3 by BQ that runs Marlin, soI don't think so.

MoonshineSG, 2016-07-12:
The scope of the plugin is to inject code nothing else. What gets injected is up to you

guysoft, 2016-07-12:
Ok,
I managed to pause at a hight and change filament using the Cura plugin pauseAtZ.py. [Its source is available here](https://github.com/smorloc/Curation/blob/master/Cura/plugins/pauseAtZ.py)

MoonshineSG, 2016-07-13:
I agree that this function should be in the slicer. Unfortunately the one I am using (Simplify3D) does not provide it (yet?). 
If the slicer can do the "pause at layer", this plugin becomes redundant and can be removed.

guysoft, 2016-07-13:
Not exactly,
A plugin could be more useful if it alerts when it reaches the layer, that
way you know when to switch

MoonshineSG, 2016-07-13:
that would be a plugin that does something when the pause gcode is hit... 

I use a much simpler way. I have a "sound" plugin that plays MP3 files. Then I add "M300 @pause"  (whcih will play pause.mp3) in the octoprint gcode script "after print job is paused" so I know when to change the filament.

guysoft, 2016-07-14:
This one?
https://github.com/MoonshineSG/OctoPrint-Sound

MoonshineSG, 2016-07-14:
yup

closed 2017-02-18 by MoonshineSG