MoonshineSG/OctoPrint-MultiColors #4: Check if injected is present

temuccio, 2016-07-13:
Hello, 
how I check if the injected change filament at layer xx is correctly present?
If I download the file and open it with text read, I can see it?
I have try but doesn't works. I need to check if the pause is present because the file are worked from many peaple.
Thanks a lot.

MoonshineSG, 2016-07-14:
Can you post the setting you use ? The GCODE and the REGEX ?

temuccio, 2016-07-14:
Hello @MoonshineSG ,
this is the GCODE:
```
G28 X Y
M117 Change filament
M0
```
and this is the REGEX:
`layer {layer},.*?`

MoonshineSG, 2016-07-14:
and inside your code there are lines that match `layer {layer},.*?` ?

temuccio, 2016-07-14:
After download the file from octoprint I don't have any line that match with `layer {layer},.*?`

MoonshineSG, 2016-07-14:
no... before that....

temuccio, 2016-07-14:
Where I check it without download the gcode?

MoonshineSG, 2016-07-14:
hmmm... you need to make sure the regex will match the line in the slicer generated gcode. you need to do this once, and if you don't change the slicer, you probably wont need to touch it again.

temuccio, 2016-07-14:
ok, but how I check if the injection are effect?

MoonshineSG, 2016-07-14:
there are 2 way. download the gcode (after you run the replacement) and check the lines ,or view it in octoprint and you will see the "pause" (head moves to 0,0) at the selected layer

closed 2017-02-18 by MoonshineSG