MoonshineSG/OctoPrint-MultiColors #5: Hardcoded path in python file __init__.py

nothing-fr, 2016-08-04:
You use hardcoded path "/home/pi/.octoprint/uploads/" in the "__init__.py" file, I'm not using octopi so i have to change it.

Can you make it changeable via a plugin configuration page ?

MoonshineSG, 2016-08-05:
sure, I'll try ...

can you tell me what does your folder is ? what OS are you using ?

MoonshineSG, 2016-08-05:
see [https://github.com/MoonshineSG/OctoPrint-MultiColors/releases/tag/1.0.3]

nothing-fr, 2016-08-05:
I'm using Octoprint on my Synology NAS. ( https://synocommunity.com/package/octoprint )

My folder for uploads is : `/volume1/octoprint/uploads`

MoonshineSG, 2016-08-05:
see if the 1.0.3 works for you

MoonshineSG, 2016-08-05:
1.0.4 is a better implementation ... let me know if it works for you ....

MoonshineSG, 2016-08-07:
@nothing-fr did you managed to run the latest version ?

nothing-fr, 2016-08-08:
Sorry I was not here this weekend, i'll try it today if i have time, thanks.

MoonshineSG, 2016-08-08:
no worries... was just curious if it works for you (it should :D )

closed 2017-02-18 by MoonshineSG