MoonshineSG/OctoPrint-MultiColors #6: MultiColors not working anymore

Brianp82, 2016-11-09:
Hey, so I downloaded MultiColors and I was using it perfectly for about a week. I just used M300 at certain layers to have the printer emit a tone so I know when its time to change my color. But yesterday I got a new pi3 and reinstalled octopi. I did everything the same. I am on Version: 1.3.0rc1 (rc/devel branch). I have the plugin installed. Now when I run the process it says the gcode was injected. But I get no tones during my print as if the mulitcolor didnt inject the code. What can I do to fix this? 

P.S. I already tried updated back to stable and then updated back to devel branch and reinstalled the plugin. Thanks

MoonshineSG, 2016-11-10:
a look at the log files might help to see what the error is

Brianp82, 2016-11-10:
This is all I can see in the log so far. No actual "error"

2016-11-09 22:51:37,701 - octoprint.filemanager.analysis - INFO - Starting analysis of local:D5_double_big_cirlce_x_10.gcode
2016-11-09 22:52:00,600 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'1 42 43', u'gcode': u'm300\nm300\nm300', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'D5_double_big_cirlce_x_10.gcode'}'
2016-11-09 22:52:00,634 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/D5_double_big_cirlce_x_10.gcode'
2016-11-09 22:52:00,638 - octoprint.plugins.multi_colors - INFO - Trying layer '1'...
2016-11-09 22:52:01,035 - octoprint.plugins.multi_colors - INFO - Trying layer '42'...
2016-11-09 22:52:01,475 - octoprint.plugins.multi_colors - INFO - Trying layer '43'...
2016-11-09 22:54:40,241 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 14, current line = 15

Brianp82, 2016-11-10:
INFO - Starting analysis of local:D5_double_big_cirlce_x_10.gcode
2016-11-09 02:18:50,337 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2016-11-09 02:18:55,120 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'42 43', u'gcode': u'm300\nm300\nm300', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'D5_double_big_cirlce_x_10.gcode'}'
2016-11-09 02:18:55,181 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/D5_double_big_cirlce_x_10.gcode'
2016-11-09 02:18:55,192 - octoprint.plugins.multi_colors - INFO - Trying layer '42'...
2016-11-09 02:18:55,617 - octoprint.plugins.multi_colors - INFO - Trying layer '43'...

Brianp82, 2016-11-10:
2016-11-09 02:16:39,721 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2016-11-09 02:16:46,072 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'32 33', u'gcode': u'm300\nm300\nm300', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'D3_Double_x_10.gcode'}'
2016-11-09 02:16:46,144 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/D3_Double_x_10.gcode'
2016-11-09 02:16:46,154 - octoprint.plugins.multi_colors - INFO - Trying layer '32'...
2016-11-09 02:16:46,485 - octoprint.plugins.multi_colors - INFO - Trying layer '33'...
2016-11-09 02:17:48,297 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2016-11-09 02:17:55,358 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'42 43', u'gcode': u'm300\nm300\nm300', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'D6_x_10.gcode'}'
2016-11-09 02:17:55,361 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/D6_x_10.gcode'
2016-11-09 02:17:55,362 - octoprint.plugins.multi_colors - INFO - Trying layer '42'...
2016-11-09 02:17:55,771 - octoprint.plugins.multi_colors - INFO - Trying layer '43'...
2016-11-09 02:18:20,646 - octoprint.filemanager.analysis - INFO - Starting analysis of local:D5_double_big_cirlce_x_10.gcode
2016-11-09 02:18:50,337 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2016-11-09 02:18:55,120 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'42 43', u'gcode': u'm300\nm300\nm300', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'D5_double_big_cirlce_x_10.gcode'}'
2016-11-09 02:18:55,181 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/D5_double_big_cirlce_x_10.gcode'

Brianp82, 2016-11-10:
Sorry for the multiple posts. This is all the stuff I can see in my log. Not sure if that is before or during printing or when I am actually injecting the gcode for multicolor. Is there something specific I can search my logs for?

MoonshineSG, 2016-11-10:
if you search in your gcode, can you find "layer XX," ? (where XX is the layer you want to switch filament )

Brianp82, 2016-11-10:
I just copied my whole log for you here. I searched for 'layer' and the only results showed up in the last few log clips I sent you.
https://gist.github.com/Brianp82/9f0222f34776e92f645b683b653ab76f

MoonshineSG, 2016-11-10:
in your gcode file are there lines that contain "layer" ? what slicer are you using ?

Brianp82, 2016-11-10:
I don't know how to check that. I am using S3D. Remember, I used the same gcode before and it worked and beeped during printing. Now it's not.

MoonshineSG, 2016-11-10:
just open a gcode generated by your slicer. 

you can send me both, the gcode before and after running the plugin and I'll try to simulate your error (unless it's a confidential gcode)

Brianp82, 2016-11-10:
I'm sorry I'm still a little new to this. What do you mean open a gcode? How am I opening it? So you want a gcode file before I process the plugin and after? Details please and I will try my best. Thanks.

MoonshineSG, 2016-11-10:
I mean visually inspect it in your favourite text editor. 

Yes, before and after you process it with the plugin. Before - as generated by S3D, after you can download it from the OctoPrint file manager (AFTER you process it)

Brianp82, 2016-11-10:
Hey, so I did what you said. And I totally see the gcode inserted at the correct layer. There is one thing I noticed. All the M300 commands have a lower case 'm'. So just for the hell of it I just re injected the code on layer 1 with M300(capital M) and I got a beep!!!! Do the gcode commands have to be capital letter? When I manually enter m300 in the terminal tab it works with m300 & M300. So I figured it doesn't matter but maybe it does for this specific plugin. I can still give you both of the gcode files if you want but that is my findings. I have tried 3 separate files before with m300 and never got a beep. So I am hoping that was the only problem but doesn't make sense since m300 worked in terminal. Hope its just not a coincidence. Thanks let me know.

MoonshineSG, 2016-11-10:
OctoPrint does the conversion and sends to printer to upper case when you type in the terminal. I guess it does not do that when it sends the files for printing. 

(in the terminal, use the up key to see your previous command. You'll notice it's in upper case :D )

Brianp82, 2016-11-10:
Here are the two file

[injected file.txt](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/582727/injected.file.txt)
[clean file.txt](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/582728/clean.file.txt)

Brianp82, 2016-11-10:
Ahhh good info. Then that has to be the issue! The plugin injects the code as lower case. I changed it to uppercase M and seems to have worked. Good to know for future reference.

MoonshineSG, 2016-11-10:
I'm glad it works. I wasn't even sure if there is anyone using the plugin :D

Brianp82, 2016-11-10:
Oh yeh me and my friend love it. We actually wanted to find a way to use the plugin so that it can send you a text/email when you reach a certain layer. That would help a lot as we can then return to our printer and change colors while it's still printing. We don't pause the print.  Also another cool feature would be to incorporate a estimated time it would take to reach a selected layer. That way you could set a timer if you wanted as well. I figured the email would be easy to do since there is a way to send an email already when the print completes. So should be impossible to do the same when a layer is reached??

MoonshineSG, 2016-11-10:
Unfortunately adding this functionality in MultiColors is kinda out of scope, but... you can use a small "hack"... 

The MultiColor plugin will insert **anything** at the requested layer. You can insert a custom "action" code that, a separate plugin will detect and do ... whatever you want it to do... 

 Note: _this might require changes to the firmware, but if you use Marlin, you can have a look at my Marlin  fork and look at lines 7237 and 8403 in "Marlin_main.cpp"... It adds a new M code `M808` that would echo back as "//action:" and parameters you give it. OctoPrint has a `octoprint.comm.protocol.action` hook that would catch it and act on it. _ 

So if you insert `M808 email layer 33` then have a plugin (see "gocode_action.py" under my "OctoPrint-Plugins" ) that sends you an email (it could even know which layer it is!)

closed 2017-02-18 by MoonshineSG