MoonshineSG/OctoPrint-MultiColors #7: Not injecting GCode

spankurmonkey, 2017-01-04:
I have tried Cura, MatterControl and S3D and have yet to get this plugin to inject the necessary gcode though the plugin says it did.... I download it from Octoprint and view in Notepad ++ and cannot find any injected code.

MoonshineSG, 2017-01-04:
please post your configuration (regex and replacement gcode) as well as a sample of your gcode.

spankurmonkey, 2017-01-04:
I want it to inject at layer 5.
`G0 F9000 X115.769 Y105.587
G1 F3000 X114.271 Y104.088 E268.35202
; LAYER:5
G0 F9000 X114.271 Y105.427 Z1.2
G0 X114.045 Y105.872
G0 X114.045 Y105.872`

[image]

MoonshineSG, 2017-01-04:
try removing the comma in the regex so it's `LAYER:(.*).*?`
If the regex doesn't match your line, it will never insert the replacement gcode

spankurmonkey, 2017-01-04:
yeah, that injects alright but at every single layer.....

MoonshineSG, 2017-01-04:
See the edited message . Sorry about that.

MoonshineSG, 2017-01-04:
Now thinking about it, it might be wrong... if you have many layers it will also insert at layer 50, 51,52,53...  that's smth you prob don't want.
Try removing the last ".*?"  part

spankurmonkey, 2017-01-04:
LAYER:(.*)   still injects at every layer.

The layer syntax is:   ; LAYER:5

MoonshineSG, 2017-01-05:
For your case, the correct regex would be `; LAYER:{layer}` 
Please note, that different slicers might require different regex...

diomark, 2017-02-09:
I need some help with this.. My layer syntax looks like this -

; ------------------------------------
; Layer #21
; ------------------------------------

...and for the life of me I can't figure out the correct regex. Any help?
Thanks, -m

MoonshineSG, 2017-02-09:
try `; Layer #{layer}`

MoonshineSG, 2017-02-09:
even better: `; Layer #{layer}$` so it does not stop at layer 210, 211, 212 ...  :D

diomark, 2017-02-09:
Hmm.. didn't work. re-downloaded the file and I can't see anything after Layer #20:(

(I tried ; Layer #{layer}$)

MoonshineSG, 2017-02-09:
if the file is not copyrighted, can you sent it to me ? or a larger part surrounding layer 20 or 21

diomark, 2017-02-09:
It's just a simple calibration cube, as generated by CraftWare. attached. 
[20mm_calibration_cube_test.txt](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/762333/20mm_calibration_cube_test.txt)

MoonshineSG, 2017-02-09:
how about the replacement string  ? 

_on a non related issue, i notice your heating bed and nozzle at the same time, but only wait for nozzle. Chances are that the bed will not reach the expected temperature by the time the print starts. And unless you have a super duper power source,  heating them both at the same time might be putting  a lot of pressure on the source._

diomark, 2017-02-09:
I'm just using your defaults right now (M117 Change filament
M0) - figured I'd get this part working before the next part. The cura pause at z was working perfectly for me, so I'd probably adopt that. (just that, I like craftware much better then cura..)

(Using the pre-heat pla), my bed usually comes up to temp a few minutes before the nozzle does. Thanks though I didn't really consider the power draw..
-m

MoonshineSG, 2017-02-09:
you're using windows to slice your model ? Seams like CraftWare inserts CRLF (windows standard) as line terminator.... `; Layer #{layer}\r` seams to do the trick...

diomark, 2017-02-09:
Figured it was something like that.. (and yes, running craftware on windows 10..)

Thanks! will test this later tonight.

closed 2017-02-18 by MoonshineSG