MoonshineSG/OctoPrint-MultiColors #8: Doesn't work with folders

heathicus, 2017-02-17:
I'm using the dev branch of Octoprint which supports organizing files in folders.  The MultiColor plugin only works on files in the root of the storage.  It fails when using a file in any folder below that.

MoonshineSG, 2017-02-17:
Yes, that is right. For now. Mostly because I don't use folders. Can't really see the use for them.
But if there seams to be a use for it, I'll look into it. Give me a few days...

heathicus, 2017-02-17:
I have a lot of different files for different projects that I print repeatedly and I use folders to keep things organized.  The MultiColor plugin is my favorite way to inject the pause code when doing multiple color changes.  It's sort of a pain to upload the file to the root, perform the injection, download the file, delete it from the root, then re-upload it to the proper folder.  First world problems, I know, but if the MultiColor plugin could support folders, it would save me a lot of time.

MoonshineSG, 2017-02-18:
as it turns out, it's a super easy fix... enjoy!

synman, 2017-02-18:
thank you so much for fixing this... I stumbled upon it a couple weeks ago and have been moving stuff back into the ~/ ever since.  No longer!

Thanks again!

Shell

closed 2017-02-18 by MoonshineSG

heathicus, 2017-02-18:
Thanks so much for the quick update!!