MoonshineSG/OctoPrint-MultiColors #10: Failed to insert code for layer

getyourbots, 2017-03-05:
I'm having an issue with the plugin, when trying to insert some layers.
I'm getting this error in the logs:

> Failed to insert code for layer

Here is the entire portion, from the logs, regarding multicolors (the _[=]_ lines were added by myself, for debugging)
> 2017-03-05 09:00:12,413 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'1 2 3', u'gcode': u'M117 Change filament\nM0', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'mp_select_mini_magic_numbers_plaque.gcode'}'
> 2017-03-05 09:00:12,416 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/mp_select_mini_magic_numbers_plaque.gcode'
> 2017-03-05 09:00:12,545 - octoprint.plugins.multi_colors - INFO - [=] IN INJECT CODE
> 2017-03-05 09:00:12,545 - octoprint.plugins.multi_colors - INFO - [=] 3
> 2017-03-05 09:00:12,546 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '1'...
> 2017-03-05 09:00:12,547 - octoprint.plugins.multi_colors - INFO - [=] Search pattern: (layer 1,.*?$)
> 2017-03-05 09:00:12,639 - octoprint.plugins.multi_colors - INFO - Failed to insert code for layer 1
> 2017-03-05 09:00:12,641 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '2'...
> 2017-03-05 09:00:12,642 - octoprint.plugins.multi_colors - INFO - [=] Search pattern: (layer 2,.*?$)
> 2017-03-05 09:00:12,735 - octoprint.plugins.multi_colors - INFO - Failed to insert code for layer 2
> 2017-03-05 09:00:12,736 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '3'...
> 2017-03-05 09:00:12,737 - octoprint.plugins.multi_colors - INFO - [=] Search pattern: (layer 3,.*?$)
> 2017-03-05 09:00:12,830 - octoprint.plugins.multi_colors - INFO - Failed to insert code for layer 3

After investigating for a bit, looks like the problem is the regex not being found in the file, which is _line 80_ in the __init__.py, respectively:

> _test = re.search(search, m)

I'm using the latest version of octoprint: _Version 1.3.1 (master branch)
multicolor plugin was installed through plugin manager of octoprint.

Thanks in advance !

MoonshineSG, 2017-03-05:
can you please post your REGEX and a sample of your file `mp_select_mini_magic_numbers_plaque.gcode` ?

getyourbots, 2017-03-05:
Sure, here's the GCODE file.
[mp_select_mini_magic_numbers_plaque.zip](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/819248/mp_select_mini_magic_numbers_plaque.zip)
I've tried others and I'm getting the same thing.

Basically I haven't changed the REGEX at all, just used the default settings/regex, put some layer number/s and hit _Process_
[image]
[image]

MoonshineSG, 2017-03-05:
ok. problem is that in your current configuration, Slic3r (thats what you're using right?)  does not insert the layer number so the plugin can't know where to insert the additional gcode. 

Good news. It's an easy fix. You need to insert `; layer [layer_num],`  in the "Layer Chance G-codes"

Of course after this you reupload the file to OctoPrint and try to insert again.

MoonshineSG, 2017-03-05:
instead of `; layer [layer_num],` you can write anything as long as you modify the REGEX to match your string

getyourbots, 2017-03-05:
Thanks a lot.
Yes, I'm using Slic3r.
Got the point with the REGEX as well.

I'm just setting it up and testing. By the way, the `; layer [layer_num]` should be in the _Before layer change G-code_ right ? There's a field for _After layer change G-code_ in Slic3r as well.

MoonshineSG, 2017-03-05:
Before.  Well... to be honest it's up to you. As long as you know where you put it and use the layer number accordingly. If you put it after, the GCODE with the pause will be inserted after the layer is finished so the printer will pause after the layer you "tagged" is completed. 

Most slicers wil insert a "now starting layer X" note. So I wrote the (minimal) documentation with that in mind, but the plugin doesn't care about it. If it finds the keywords, it adds the additional code. 

Hope this makes sense.

getyourbots, 2017-03-05:
Yes, it does perfectly.
I was thinking to give it the 1st layer which I want with the new color, so that will be before though.
But in my opinion it's cool you can do it both ways (before & after, based on how you want it).

Once again, thanks and I appreciate your fast response, especially in weekend.
Have a nice one !

closed 2017-03-05 by getyourbots

jstuewe, 2021-02-01:
I'm using PrusaSlicer on my Ender 3 R2 and had to play around with the REGEX expression.
This slicer puts the layer comment on the line with the layer change Gcode.
ie.  "G1 Z1.000 F9000.000 ; move to next layer (4)"

To get this slicer to work the correct REGEX expression is:

move.to.next.layer.\({layer}\)