MoonshineSG/OctoPrint-MultiColors #11: Add option to inject gcode before regex match

jewzaam, 2017-03-26:
I use Slic3r and layer comments are on the line for moving to next layer's first position.  To do layer change, need to be able to inject gcode before this line.

Example
```
G1 Z0.840 F9600.000 ; move to next layer (3)
```

Current plugin functionality:
```
G1 Z0.840 F9600.000 ; move to next layer (3)
; multi color
M117 Change filament
G1 X203 Z40
M0
```

Desired plugin functionality:
```
; multi color
M117 Change filament
G1 X203 Z40
M0
G1 Z0.840 F9600.000 ; move to next layer (3)
```

jewzaam, 2017-03-27:
Possibly just a lack of documentation and personal understanding of how regex works in python.  Digging into things and will post a PR with some doc updates if that's the case.

MoonshineSG, 2017-03-27:
It does support multi line for the replacement.  And I am not sure why the "before" option does not work...

jewzaam, 2017-03-27:
I don't see but the regex and gcode in the plugin.  But assuming the regex works the way I think it should, it can do what I need already.  Have a print in progress but will play around after that.

MoonshineSG, 2017-03-27:
I'm sure that by playing with the inserted GCODE you can make it work even if it's inserted after the matching line. 
Let me know if you need any help.

jewzaam, 2017-03-27:
True, replaying a few things got it working (at least looks good in an editor).  Testing will be of the gcode, not the plugin.  Thanks!

closed 2017-03-27 by jewzaam

jewzaam, 2017-03-27:
FWIW, my settings: https://gist.github.com/jewzaam/9f43dd0607ffb39a23967366121d1174