MoonshineSG/OctoPrint-MultiColors #13: Failed to inject gcode

WVUTampaAlum, 2017-05-07:
Hello.  I am running OctoPrint 1.3.2 and installed multi color through gui plugin manager.  When I try to use it I get inject failure.  Below is info from octoprint log file.  Also note I uninstalled the plugin and tried installing it manually which failed. 

2017-05-07 11:58:18,234 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2017-05-07 11:59:22,956 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'2', u'gcode': 
u'M117 Change filament\nM0', u'find_string': u'layer {layer},.*?', u'command': u'process', u'file': u'Main_Drain_Difuser.gcode'}'
2017-05-07 11:59:22,961 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/Main_Drain_Difuser.g
code'
2017-05-07 11:59:23,006 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '2'...
2017-05-07 11:59:23,055 - octoprint.plugins.multi_colors - INFO - Failed to insert code for layer 2

Manual Install Fail Log:

pip install "https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip"
Downloading/unpacking https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip
  Downloading master.zip (158Kb): 158Kb downloaded
  Running setup.py egg_info for package from https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
    Complete output from command python setup.py egg_info:
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?

----------------------------------------
Command python setup.py egg_info failed with error code 255 in /tmp/pip-dKUo1o-build
Storing complete log in /home/pi/.pip/pip.log

MoonshineSG, 2017-05-07:
A sample of your gcode would help ...

WVUTampaAlum, 2017-05-07:
Here is a link to gcode file on google drive.  I tried 3 different files and all failed with same error.

https://drive.google.com/open?id=0BwjRr8SEwxv9SVRLLXhXSkNrQTQ

MoonshineSG, 2017-05-08:
The regular expression need to match the line that your slicer inserts to identify the layer change. In your case is `;LAYER:` followed by the layer number so the regular expression to try is  `;LAYER:{layer}`

WVUTampaAlum, 2017-05-08:
It's working perfect.  Thank you much.  Sorry to bother you only I didn't find any documentation earlier when trying to research my problem.  Also if you had not already noticed while looking at the gcode file I am using the newest cura 2.5 slicer.

closed 2017-05-08 by MoonshineSG