MoonshineSG/OctoPrint-MultiColors #14: Layer index in GUI start at 1; in GCODE starts at 0

kroyster, 2017-05-17:
I'm using Slicer v1.2.8 to generate my GCODE in OctoPrint, and MultiColors v1.0.6 for adding pauses to swap filaments.  When I use the built-in GCODE Viewer in OctoPrint to view each layer and figure out where to add pauses, the GUI reports N layers and shows them each as layers 1 through N.  However, the GCODE starts with a layer index of 0, so it goes 0 through N-1. 

This causes each injected layer to occur 1 layer too late. I realize the fix could be accomplished in OctoPrint (the GUI), Slicer (the GCODE), or MultiColors (modified GCODE). I'm not sure which makes the most sense to report this to, but thought I'd start here.  Thanks!!

MoonshineSG, 2017-05-18:
well... since you know this happens... the easiest way would be to tell MultiCOlor to insert the code at N-1 (where N is the layer where you want the change to happen as read from the build in viewer)...

kroyster, 2017-05-18:
Ummm... yeah, I did exactly that. But then everyone else would have to learn this lesson by trial and error and make the same kludgy adjustment.  Sorry, was just trying to help others...  At any rate, thanks for this plugin - I enjoy it.

MoonshineSG, 2017-05-18:
you're most welcome!

As for the "problem", I really don't think there's a change that any of the 3 players can implement to cater for all the scenarios. For example: MultiColor can do a check and if the first layer is 0 not 1 it automatically subtracts 1 from each layer that the users inputs. But is this what the user wants ? There's no way to know... 
Oh... if all the slicers would be consistent and start with layer 1, then yes, problem solved... But that is unlikely to happen,.. so ya.. trial and error like the rest of us mortals :D 

enjoy the plugin!!

closed 2017-08-07 by MoonshineSG