MoonshineSG/OctoPrint-MultiColors #15: doesn't work with cura gcode

albebert, 2017-07-19:
Hello, 

i use cura in repetier-host for slicing.

layer in gcode are : 

;LAYER:10

so i try to user this regexp : 
.*LAYER:{layer}.*

but no luck :/

can you help me ?

MoonshineSG, 2017-07-20:
try `;LAYER:{layer}`

albebert, 2017-07-20:
same thing :(

MoonshineSG, 2017-07-20:
Send me a bigger sample

albebert, 2017-07-20:
http://dl.free.fr/ry7Xc8hHn

MoonshineSG, 2017-07-21:
the `;LAYER:{layer}` should work. i tested against your file and it's a match... 

see https://regex101.com/r/Rj4ffA/1

the `(`, `$)` are added by the plugin (https://github.com/MoonshineSG/OctoPrint-MultiColors/blob/master/octoprint_multi_colors/__init__.py#L74)

Can you paste the logs from your OctoPrint ?

MoonshineSG, 2017-07-21:
Octoprint logs? Under the settings or via SSH?

albebert, 2017-07-21:
/home/pi/.octoprint/logs/octoprint.log i only have this : 
2017-07-20 15:43:58,031 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'5', u'gcode': u'M117 Change filament\nM0', u'find_string': u';LAYER:{layer}', u'command': u'process', u'file': u'20mm_calibration_cube.gcode'}'
2017-07-20 15:43:58,042 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/20mm_calibration_cube.gcode'
2017-07-20 15:43:58,086 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '5'...

MoonshineSG, 2017-07-21:
anything else after the line  "Trying to insert multi color code for layer '5'..." ?

MoonshineSG, 2017-07-22:
`;LAYER:{layer}\r` will work. Sliced  on Windows by any chance ?

MoonshineSG, 2017-07-22:
i'm updating the plugin to handle automatically the cross platform line terminator

MoonshineSG, 2017-07-22:
update to 1.0.7 and use `;LAYER:{layer}`

albebert, 2017-07-22:
Yes slicing on windows (repetier-host). 

updated, and now works ! thanks !!

you can close :)

if i can just make a comment on your plugins, it can be great if you can pickup a file from file list instead of using the loaded file. because for loading file, printed must be UP. and add a save as fonctionnaly could be great to. .

another thanks for your great job, have a good day

closed 2017-08-07 by MoonshineSG