MoonshineSG/OctoPrint-MultiColors #16: Print fails at color change

ScottyMakesStuff, 2017-08-05:
I'm using the filament change plugin not to change filament but to pause the print so I can insert a nut into the print. 
I just use the z up button to move the print head out of the way and after move it back down the same amount of button clicks. I've tried with and without filament retraction and the print still easily snaps apart at that layer. 
It's got me baffled. I'm wondering if the z buttons are slightly inaccurate? Do I need to hand write a retraction distance and then return?

MoonshineSG, 2017-08-05:
The plugin  inserts whatever you tell it to insert and it is dependent on your particular printer

What z buttons are you talking about?

ScottyMakesStuff, 2017-08-05:
The Z buttons are for up and down on the Octoprint 'Control' tab.
It's just a simple pause at a specific layer height. It pauses and resumes nicely. Just snaps apart like there's a layer missing.
The g code checks out - the layer height hasn't skipped a layer. The only thing I can think of is the z buttons. I'll just type in a move up before the pause and then back down at the resume but I thought it worth your attention.

MoonshineSG, 2017-08-05:
I see... I suggest you don't move the Z. Most printers would have a small (but obviously significant) backlash. 

When changing filament, I move to a different X/Y position. I have a "After print job is paused" scripts set to 

```
G1 X0 Y0 F8000 ; go home
M72 @pause ; play the pause tune
```

change the filament and when I hit resume the print continues. Even if X and Y have some backslash, it will only show as a dot on the final print. Try and maybe works better for you. 

An added benefit is that you have full access to the print as the head has move away. 

The above assumes your model is sliced with "absolute" values, otherwise it doesnt work.

ScottyMakesStuff, 2017-08-05:
Yep, good advice... Thanks for the help MoonshineSG!

closed 2017-08-05 by MoonshineSG