MoonshineSG/OctoPrint-MultiColors #17: Add option to rename and/or duplicate file upon injecting color change gcode

Burritobun, 2017-11-03:
Just as the title says, allow renaming of the changed files so they aren't lost in the file menu with the same name. Also allow duplicate files to be created instead of changing the original file.

MoonshineSG, 2017-11-15:
updated with duplicate option. enjoy.

Burritobun, 2017-11-15:
Thanks to updating a 2 year old plugin! It does seem that it updates, but it removes the plugin from my server, and installing it from the plugin browsing tool in octoprint causes it to reinstall after saying that it's already installed.

MoonshineSG, 2017-11-15:
that's strange... maybe remove it and reinstall it ?

Burritobun, 2017-11-15:
There's nothing to uninstall. Doesn't show up until I try to install over it. What would be the SSH command to remove it manually?

Burritobun, 2017-11-15:
`Installing plugin "Multi Colors" from https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip...
/home/pi/oprint/bin/python -m pip install https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip
Collecting https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip
Downloading https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip (145kB)
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-MultiColors==1.0.8 from https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg (from OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: pyserial<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: requests<2.8,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.8)
[... cut ...]
/home/pi/oprint/bin/python -m pip install https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip --ignore-installed --force-reinstall --no-deps
Collecting https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip
Downloading https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/master.zip (145kB)
Installing collected packages: OctoPrint-MultiColors
Running setup.py install for OctoPrint-MultiColors: started
Running setup.py install for OctoPrint-MultiColors: finished with status 'done'
Successfully installed OctoPrint-MultiColors-1.0.8
Done!`

MoonshineSG, 2017-11-15:
hold on. found a bug... I'll have a quick fix in a short while

Burritobun, 2017-11-15:
Alright, thank you!

MoonshineSG, 2017-11-15:
try updating to 1.0.9. Sorry about that.

Burritobun, 2017-11-15:
Shows up perfect now, thanks!

closed 2017-11-15 by MoonshineSG