MoonshineSG/OctoPrint-MultiColors #18: \r required for multi line injection.

cwarts, 2017-12-19:
Injected gcode doesn't seem to apply newlines.  Using regex "LAYER:{layer}" and layer 36 with the injected code:
`G28 Y0 X0 ; home the X and Y axis 
G1 Y120; move plate forward 
M300 ; ring the buzzer 
M25 ; pause the print so you can change the filament 
G92 E0 ; reset the extruder origin 
G28 Y0 X ; home the X and Y axis 
; end multi-color `
results in
`;LAYER:36; multi colorG28 Y0 X0 ; home the X and Y axis G1 Y120; move plate forward M300 ; ring the buzzer M25 ; pause the print so you can change the filament G92 E0 ; reset the extruder origin G28 Y0 X ; home the X and Y axis ; end multi-color `

Adding /n to the code to be injected didn't seem to work, but /r did, so the new code:
`\r
G28 Y0 X0 ; home the X and Y axis \r
G1 Y120; move plate forward \r
M300 ; ring the buzzer \r
M25 ; pause the print so you can change the filament \r
G92 E0 ; reset the extruder origin \r
G28 Y0 X ; home the X and Y axis \r
; end multi-color \r`
Results in:
`;LAYER:36

; multi color

G28 Y0 X0 ; home the X and Y axis 
G1 Y120; move plate forward 
M300 ; ring the buzzer 
M25 ; pause the print so you can change the filament 
G92 E0 ; reset the extruder origin 
G28 Y0 X ; home the X and Y axis 
; end multi-color `

Thanks.

MoonshineSG, 2017-12-29:
what OS are you on ?

cwarts, 2018-02-15:
Sorry, didn't see you replied.  Linux.

tcurdt, 2018-09-09:
Is this still an issue?

MoonshineSG, 2018-09-09:
There are some isolated cases. If you have an issue, describe it and I’ll try to help