MoonshineSG/OctoPrint-MultiColors #19: Injecting gcode failed

Burritobun, 2018-03-23:
I have the plugin with default config settings, after updating to 1.3.6 I have been getting the error "Injecting GCODE failed. Replaced 0 out of 2 needed." after attempting to inject into a file. I tried editing the original along with trying to duplicate the file.

Burritobun, 2018-03-23:
Solved it by changing the "look for" to match what cura was spitting out. Cura specifies layers in this way   ";LAYER:34" so I have mine like this "LAYER:{layer}". Perhaps adding a broader search or a slicer selection button to change what the plugin looks for.

MoonshineSG, 2018-03-23:
yes, you are supposed to configure the  regular expression  to match whatever the slicer inserts... thats the whole point of having a regular expression

Burritobun, 2018-03-23:
My apologies. I would delete this thread if I knew how... Very new to this
whole github thing.

MoonshineSG, 2018-03-23:
that's alright, no harm done. we all started at some point somewhere. 

Let me know if you have other questions.

closed 2018-03-23 by MoonshineSG

Jelenbl, 2021-10-14:
Try this as your regex statement.

.([LAYER])\w+.*([layer]).