MoonshineSG/OctoPrint-MultiColors #20: ERROR: Injecting GCODE failed. Replaced 0 out of 1 needed.

TehEmoGurl, 2018-04-07:
Not sure what i'm doing wrong

Layers: "254" (Also tried "254,")

Duplicate File: Enabled

All options under "Advanced" have been left as default as i have no idea what im doing.

whenever i click "Process" it give me the error: "Injecting GCODE failed. Replaced 0 out of 1 needed."

I also tried adding 2 layers seperated by a comma or a space n still wouldn't work.

Slicer used to generate the original gCode file is Cura (Not at the computer right now but it's whatever version the latest is as i always keep it up to date and the gCode file was generated about an hour ago when i first started trying to get this plugin working to try).

Printer is Anet A6 baught mid last year. The firmware is the original default firmware it came with (I think it's marlin but i don't actually know as i've never checked). All settings on the printer itself have been left at factory defaults.

Thanks in advance for any help provided

~Jess~

[image]

icemanfred, 2018-05-06:
i ahve this same problen with robo r1+

MoonshineSG, 2018-05-06:
Depending on the slicer you use, you need to change the regular expression.

TehEmoGurl, 2018-05-06:
I tried this also but still got the same error. Not sure of the syntax. I'm using Cura and the gcode shows the layer expression as ;layer:

MoonshineSG, 2018-05-06:
Post a sample of your gcode and i’ll try to give you a working regex

TehEmoGurl, 2018-05-06:
Thank you very much. Will send the code soon as I'm home :)

majda107, 2018-05-06:
It doesn't work for me too :(

TehEmoGurl, 2018-05-06:
So here is the settings i tried:
http://prntscr.com/jegcet

And here is a sample gCode from my Cura (Version 3.2.1):
`;FLAVOR:Marlin
;TIME:4256
;Filament used: 4.54273m
;Layer height: 0.15
;Generated with Cura_SteamEngine 3.2.1
M190 S50
M104 S200
M109 S200
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
;LAYER_COUNT:111
;LAYER:-6
M107
G0 F3600 X94.259 Y59.78 Z0.24
G1 F1800 X94.601 Y59.26 E0.04968
G1 X94.75 Y59.053 E0.07004
G1 X95.134 Y58.569 E0.11936
G1 X95.267 Y58.415 E0.1356
G1 X95.721 Y57.937 E0.18822
G1 X95.873 Y57.791 E0.20505
G1 X96.35 Y57.372 E0.25573
G1 X96.539 Y57.22 E0.27509
G1 X97.039 Y56.855 E0.3245
G1 X97.228 Y56.729 E0.34264
G1 X97.796 Y56.387 E0.39556
G1 X97.947 Y56.305 E0.40928
G1 X98.536 Y56.02 E0.46151
G1 X98.757 Y55.923 E0.48077
G1 X99.344 Y55.7 E0.5309
G1 X99.553 Y55.631 E0.54847
G1 X100.143 Y55.465 E0.59739
G1 X100.394 Y55.407 E0.61796
G1 X101.012 Y55.291 E0.66815
G1 X101.214 Y55.263 E0.68443
G1 X101.908 Y55.197 E0.74008
G1 X102.439 Y55.179 E0.78249
G1 X105.228 Y55.178 E1.00512
G1 X106.058 Y55.213 E1.07143
G1 X106.967 Y55.283 E1.14421
G1 X107.502 Y55.259 E1.18695
G1 X108.154 Y55.197 E1.23923
G1 X108.688 Y55.179 E1.28188
G1 X111.362 Y55.178 E1.49534
G1 X112.193 Y55.213 E1.56173
G1 X113.096 Y55.283 E1.63403
G1 X113.636 Y55.259 E1.67717
G1 X114.287 Y55.197 E1.72937
G1 X114.823 Y55.179 E1.77218
G1 X117.612 Y55.178 E1.99481
G1 X118.44 Y55.213 E2.06097
G1 X118.655 Y55.231 E2.07819
G1 X119.29 Y55.317 E2.12934
G1 X119.505 Y55.356 E2.14678
G1 X120.125 Y55.497 E2.19754
G1 X120.331 Y55.553 E2.21458
G1 X120.961 Y55.757 E2.26744
G1 X121.142 Y55.824 E2.28285
G1 X121.74 Y56.079 E2.33474
G1 X121.934 Y56.172 E2.35191
G1 X122.511 Y56.481 E2.40416
G1 X122.687 Y56.586 E2.42052
[... 2207 lines cut ...]
G1 X103.876 Y69.935 E1459.92789
G1 X103.876 Y70.501 E1459.95222
G1 X103.876 Y71.066 E1459.97651
G1 X103.876 Y71.632 E1460.00084
G1 X103.876 Y72.198 E1460.02517
G1 X103.876 Y72.763 E1460.04946
G1 X103.876 Y73.329 E1460.07375
G1 X103.876 Y73.894 E1460.09799
G1 X103.876 Y74.46 E1460.12227
G1 X103.876 Y75.026 E1460.14656
G1 X103.876 Y75.591 E1460.1708
G1 X103.876 Y76.157 E1460.19508
G1 X103.876 Y76.723 E1460.21937
G1 X103.684 Y77.096 E1460.23461
G0 F3600 X103.618 Y77.03
G0 X104.677 Y77.941
G0 X115.463 Y77.941
G0 X115.9 Y77.665
G0 X115.97 Y77.725
;TIME_ELAPSED:744.325061
;LAYER:1`

Thanks for the help

--Jess--

MoonshineSG, 2018-05-07:
try regex `;LAYER:{layer}`

TehEmoGurl, 2018-05-07:
Thanks Moon, will test it on mine after my current print finishes (Estimated 10 hours) n let you know if it works for me or not :)

MoonshineSG, 2018-05-08:
@majda107 : your code might be different.  paste a sample (only the part where the layer changes 
Example:

```
[...]
G0 F3600 X124.671 Y58.194 
;TIME_ELAPSED:87.310460
;LAYER:-5 
M140 S50 
G0 X124.671 Y58.194 Z0.464 
;TYPE:SUPPORT 
G1 F1800 X124.539 Y58.058 E195.70068 
[...]
```

TehEmoGurl, 2018-05-08:
Thank you very much @MoonshineSG ! Just tried the regex you gave me and it successfully injected ^.^

@majda107 The same regex should work for you assuming that layer number iterations in Curas gCode generator is enabled by default (I assume it is as i have looked everywhere in the preferences section and cannot find a way to disable it).

If you want to be sure of this just open up any gCode file sliced in Cura with a text editor (I highly recommend **NotePad++** if you don't already have it).

If using **Notepad++** go to **Search** at the top then **Find** (Or press **Ctrl+F**). You can do this in windows NotePad and most other basic text editors too.

In the search field type 

> ;Layer:

 and click **Find** / **Find Next**.

Assuming you were successful in finding a layer iteration the settings in the screenshot below should fix your injection issues.

**NOTE: Regex IS case SeNsItIvE!**

[image]

**NOTE: The LAYERS field should have ONLY the numbers inside the "quotes" with either spaces or commas without "quotes" around them.**

**WARNING: _The Add GCODE in the example should NOT be used. I don't believe i have made a working code to achieve successful realignment of layers after changing filament here and the nozzle would likely smash into the model._**

@MoonshineSG Am i correct in that thought? And if so, how would one go about fixing such an issue?

Would this be a suitable solution to force the nozzle to always raise above the model and then come down on top to the correct location as it continues from where it left off?

```
G28
M300
M117 Change filament
M0
G28
G0 Z200
```

**Note: I'm using the Anet A6 which has a max Z height of 250mm and a gantry of approx 55mm.**

Thanks again, hugely appreciate the help. Can't wait to do some awesome multi coloured prints! ^.^

--Jess--

majda107, 2018-05-08:
Here is my part of GCode where is layer change!

[image]

TehEmoGurl, 2018-05-08:
@majda107 Could you show me a screenshot of you multi color plugin settings?

majda107, 2018-05-08:
@TehEmoGurl I'd like to, but I am printing RN and my raspberry isn't connected, my printer will finish after few minutes, I'll upload it then!

MoonshineSG, 2018-05-08:
@TehEmoGurl : NOOOOO!!!! G28 is not to be used in the middle of your print - no printer (affordable) would be able to do a homing at 100% accuracy and without hiting you already half prited model!!!!!

All you need is to Pause the print and preferable move the head away from the model so that the hot end soend cause a "bulge". `M0` usually pauses and moves the head to X0/Y0 (at least on Marlin)

If you're not sure, just take any model, slice it, add a "change color" at an early layer (2 woul dbe a good choice) and print it. Basically test that your printer does on M0 . (if adventurous, read the firmware source code! ) 

 @majda107: based on your sample the `;LAYER:{layer}` should work ust fine... What is the error ? Maybe you can have a look at the logs and see if there's any error ?

TehEmoGurl, 2018-05-08:
@MoonshineSG Hmmmmm, interesting. I actually did it this way when i originally couldn't get our plugin working. i manually edited the gCode and i did indeed accidentally smash my X Carriage into the part. i did an emergency stop, G28 to re-home again, then moved the Z up to 100mm and then told it to continue and it aligned everything just fine.

Here's a photo of the print:

[image]

The reason i prefer to home all axis before changing the filament and then again after then raise the Z above the location it needs to go to is i find it very easy to accidentally knock the X axis out of alignment or accidentally push down too hard trying to get the filament into the top of the extruder as it's very difficult to align it through the gears. If i force it to home before and after then anything i screwed up during the filament change is fixed.

Also note that the other day i tried the built in "Change Filament" option in the "Tune" section of the built in controls. The printer auto-homed all axis, auto-ejected the filament, then i reloaded the filament and told it to continue and i was surprised to find that it was completely out of alignment. I quickly paused it, re-homed manually then told it to continue and it was still out of alignment. the only way i could get it even close was to disable the steppers and try to manually align the X and Y.

As you can probably guess i failed to do this, however i was only off by about 0.5mm, where-as the auto change filament alignment was off by about 3-4mm.

Human 1 - 0 Machine

If i shouldn't use this method, can u recommend a good way to have it automatically re-align everything somehow before it continues printing in case i knock something out of alignment when changing the filament?

If not don't worry about it. I'm going to do some more experimenting with the G28 either way and see how many times out of 10 i can get it to auto-align correctly since I've only ever done it the 1 time and might have just got really damn lucky xD

MoonshineSG, 2018-05-08:
if using marlin, there's a way to NOT power down the stepper motors when the printer pauses (I'm not sure if this can be achieved by M commands or a firmware build is required. See `DEFAULT_STEPPER_DEACTIVE_TIME`, `DISABLE_X`, etc but those should be False by default ). In that way, changing the fillament would not knock the head out of the position (shoudl be 0x0y) and will  be able to safely and accuratly return to the paused position.

TehEmoGurl, 2018-05-08:
@MoonshineSG Thanks for the info, will look into that! :)

MoonshineSG, 2018-05-08:
"If i shouldn't use this method, can u recommend a good way to have it automatically re-align everything somehow before it continues printing in case i knock something out of alignment when changing the filament?"

nope! we've all been there - for whatever reason -  and using "visuals" does not produce pretty results :D

MoonshineSG, 2018-05-08:
@TehEmoGurl , sorry should be "DISABLE_INACTIVE_"... and it's in `Configuration_adv.h`. Check out `M18` and `M84` 

```
// Default stepper release if idle. Set to 0 to deactivate.
// Steppers will shut down DEFAULT_STEPPER_DEACTIVE_TIME seconds after the last move when DISABLE_INACTIVE_? is true.
// Time can be set by M18 and M84.
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // set to false if the nozzle will fall down on your printed part when print has finished.
#define DISABLE_INACTIVE_E true
```

chadsittig, 2018-09-09:
I'm getting the same error. The gcode was generated with Slic3r:
--------
G1 X87.989 Y133.455 E0.03487
M204 S1000
;BEFORE_LAYER_CHANGE
G92 E0.0
;3.2

G1 F8640
G1 X87.262 Y132.729 E-0.23727
G1 X86.677 Y132.523 E-0.14315
G1 X86.677 Y132.721 E-0.04570
G1 X87.700 Y133.743 E-0.33388
G1 E-0.04000 F2100.00000
G1 Z3.600 F10800.000
;AFTER_LAYER_CHANGE
;3.2
G1 X88.137 Y132.375
----------

I've tried the following for regular expression:
;BEFORE_LAYER_CHANGE
;AFTER_LAYER_CHANGE
;BEFORE_LAYER_CHANGE\n;[layer_z]
...and various other combinations. The error is the same "Injecting GCODE failed. Replaced 0 out of 1 needed"

MoonshineSG, 2018-09-10:
you might have a "funny" char at position 155565. Try to look for it and see what it is ? 
Any comments in the GCODE that are no english ?

chadsittig, 2018-09-10:
Thanks for the quick reply.

Character in position 155565 is a "0".

I found the problem:

> ; filament_notes = "List of manufacturers tested with standart PET print settings for MK2:\n\nE3D Edge\nFillamentum CPE GH100\nPlasty MLADEČ PETG"<

I'm able to inject the GCODE using ";BEFORE_LAYER_CHANGE", however the filament supplier "MLADEC" is generated from slic3r.

I believe this is the same issue as posted here:
[https://github.com/MoonshineSG/OctoPrint-MultiColors/issues/27](url)

MoonshineSG, 2018-09-10:
yes... I'm trying to fix it to survive foreign charcters :D

MoonshineSG, 2018-09-10:
a new relaease with a fix that shoudl be able to handle non latin characters is now available

closed 2018-09-10 by MoonshineSG

Jelenbl, 2021-10-14:
Try this as your regex statement.

.([LAYER])\w+.*([layer]).