MoonshineSG/OctoPrint-MultiColors #21: Blank line above inserted code

upD8R, 2018-05-06:
Hi there,

if I insert a multi line code block with my own regular expression (to match my gcode) everything runs well except there is a blank line directly above the inserted code:

`M117 INDICATOR-Layer28

; multi color
;------------------------------
; PAUSE & PRESENT BED 
;------------------------------
;`

I don't know what causes this and I don't have a clue how printers deal with blank lines in gcode. Maybe it's not an issue at all or dependent on FW?

MoonshineSG, 2018-05-06:
No need to worry about how FW treats is. OctoPrint doesn’t send blank lines! ;)
Enjoy!

upD8R, 2018-05-06:
Excellent, thanks for the quick response and your plugin!

closed 2018-05-06 by MoonshineSG