MoonshineSG/OctoPrint-MultiColors #23: v1.0.12 update loop

TehEmoGurl, 2018-05-08:
Just tried updating to v1.0.12 but after it says it successfully updated and reboots the server it says the v1.0.12 update is avilable in the toast bubble in the top right again.

I have checked the plugin manager and confirmed that the version running is v1.0.11

Here is the output from the update window:

`Updating, please wait.
Collecting https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/1.0.12.zip
Downloading https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/1.0.12.zip (145kB)
Requirement already satisfied (use --upgrade to upgrade): OctoPrint-MultiColors==1.0.11 from https://github.com/MoonshineSG/OctoPrint-MultiColors/archive/1.0.12.zip in /home/pi/oprint/lib/python2.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: werkzeug<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/Werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: sockjs-tornado<1.1,>=1.0.3 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: markdown<2.7,>=2.6.4 in /home/pi/oprint/lib/python2.7/site-packages/Markdown-2.6.8-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: pyserial<3.5,>=2.7 in /home/pi/oprint/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: netaddr<0.8,>=0.7.17 in /home/pi/oprint/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: watchdog<0.9,>=0.8.3 in /home/pi/oprint/lib/python2.7/site-packages/watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: pylru<1.1,>=1.0.9 in /home/pi/oprint/lib/python2.7/site-packages/pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: requests<3,>=2.18.4 in /home/pi/oprint/lib/python2.7/site-packages/requests-2.18.4-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: semantic_version<2.5,>=2.4.2 in /home/pi/oprint/lib/python2.7/site-packages/semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: psutil==5.4.3 in /home/pi/oprint/lib/python2.7/site-packages/psutil-5.4.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Click<6.3,>=6.2 in /home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/pi/oprint/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: websocket-client<0.41,>=0.40 in /home/pi/oprint/lib/python2.7/site-packages/websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: python-dateutil<2.7,>=2.6 in /home/pi/oprint/lib/python2.7/site-packages/python_dateutil-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: wrapt<1.11,>=1.10.10 in /home/pi/oprint/lib/python2.7/site-packages/wrapt-1.10.11-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: futures<3.2,>=3.1.1 in /home/pi/oprint/lib/python2.7/site-packages/futures-3.1.1-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: emoji<0.5,>=0.4.5 in /home/pi/oprint/lib/python2.7/site-packages/emoji-0.4.5-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: monotonic<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/monotonic-1.3-py2.7.egg (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: itsdangerous>=0.21 in /home/pi/oprint/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: MarkupSafe in /home/pi/oprint/lib/python2.7/site-packages/MarkupSafe-1.0-py2.7-linux-armv7l.egg (from Jinja2<2.9,>=2.8->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: certifi in /home/pi/oprint/lib/python2.7/site-packages/certifi-2017.04.17-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: backports.ssl_match_hostname in /home/pi/oprint/lib/python2.7/site-packages/backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: blinker in /home/pi/oprint/lib/python2.7/site-packages/blinker-1.4-py2.7.egg (from Flask-Principal<0.4,>=0.3.5->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: Babel>=1.0 in /home/pi/oprint/lib/python2.7/site-packages/Babel-2.4.0-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: speaklater>=1.2 in /home/pi/oprint/lib/python2.7/site-packages/speaklater-1.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-MultiColors==1.0.11)
Requirement already satisfied: webassets>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/webassets-0.12.1-py2.7.egg (from Flask-Assets<0.11,>=0.10->OctoPrint->OctoPrint-MultiColors==1.0.11)
[... cut ...]
Installing collected packages: OctoPrint-MultiColors
Running setup.py install for OctoPrint-MultiColors: started
Running setup.py install for OctoPrint-MultiColors: finished with status 'done'
Successfully installed OctoPrint-MultiColors-1.0.11
You are using pip version 9.0.3, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
The update finished successfully and the server will now be restarted.`

Let me know where to find other logs that may help with fixing the issue. thanks :)

MoonshineSG, 2018-05-08:
I’m an idiot. Forgot to update version. Try again and it goes up to 0.13 which should move out of the loop

TehEmoGurl, 2018-05-08:
Great job, works now and confirmed correct v1.0.13 in plugin manager :)

closed 2018-05-08 by TehEmoGurl