MoonshineSG/OctoPrint-MultiColors #26: Trying to create a good Extrusion code(not an issue)

IceflareTech, 2018-05-12:
This isn't an issue with the plugin at all it works great and I love it! 
I am trying to create a script to change filament without zeroing axis and ran in to an issue. I can get the extruder position with M114 but dont know how to cast that in to a variable to use. In the below example I have to manually input the E value from the last line in layer 1, which makes multiple colors a pain. Here is an example with how I do it now and with where I think I need help. Pay no attention to actual values as I need to tweak extrude lengths for a color change, retracts etc.
--------------------------------------------------------------------------------
G1 X104.201 Y138.980 E374.79387
;LAYER:2

; multi color
<-----------------This is where I would load a variable with M114's E value
G91                       ;Set to absolute positioning
G1 Z20 F200 E-3           ;retract 3mm of feed stock move 20
G90                       ;Set to relative positioning
G1 Y-200                  ;Move bed back
M117 Change Filament...   ;Display message
G91                       ;Set to absolute positioning
G1 F200 E-80              ;retract 80mm of feed stock
M0                        ;Wait for buton press( is a resume in octoprint)
M117 Extruding...         ;Display message
G1 F200 E100              ;extrude 100mm of feed stock
M117 Press to return to print..         ;Display message
M0                    ;Wait for buton press( is a resume in octoprint)
G4 P3000                  ;Wait 3 secounds
G1 F200 E3                ;extrude 3mm of feed stock
G92 E374.79387            ;Set E back to last reading <-----This is where I would need to use M114 variable
G4 P300                   ;Wait 1/3 secound
G90                       ;Set back to relative positioning
------------------------------------------------------------------------------
Can you run C alongside or is there a way in M/G code to save that data and point to it? Any help is appreciated from anyone, and THANKS for the great plugin!

MoonshineSG, 2018-05-12:
you can't. at least not easily. modify the firmware to add a "save" and "read" codes would be the best, yet not the easiest. 

but the question you should ask is "why?". Why do you need to do a zeroing of axes ? 

the way (as far as my printer goes - i run Marlin) M0 works: move to 0x/0y, pause. 
when you resume, it ust continues the gcode which (unless you use relative coordinates - which AFIK is not very common) will continue exactly from where you stoped.  Why zero the axes ?

upD8R, 2018-05-12:
Here's my code (not actually mine but found on Thingiverse) if it helps you. I also zero X ad Y in case I wasn't careful enough while changing the Filament.
Multi Colors embedded Code:

> ;------------------------------
> ; PAUSE & PRESENT BED 
> ;------------------------------
> ;
> G92 E0
> G1 E-15 F300 ;retract
> G92 E0
> ;
> G91;relative
> G0 Z10 F9000 ;UP
> G90; absolute
> G0 X0 Y200 F9000 ;arbitrary absolute pos
> ;
> M300 S500 P2000 ;BEEP
> M25 ; PAUSE
> ;

Resume code in Octoprint:

> M300 S500 P2000 ;BEEP
> G28 X Y ;home XY ONLY, to negate any unpowered bed movement by user, only as accurate as the endstops
> G90; absolute
> G0 X0 Y200 F9000 ;arbitrary absolute pos
> G92 E0
> G1 E15 F300 ;extrude
> G92 E0
> G91;relative
> G0 Z-10 F9000 ;down
> G90; absolute

MoonshineSG, 2018-05-12:
G28 is never 100% accurate. and the Z+10 -10 is unlikely to get you back to EXACTLY the same Z (unless you have a super awsome printer)

I would stick to my "just be carefull" aproach and not mess with homming. :D

upD8R, 2018-05-12:
Agreed and before using your plugin I used the Filament sensor on my printer to force a pause. This is also moving on Z then. In my environment the results with your plugin and the code above are more than sufficient for my needs. Couldn't see any offset or similar things.
[image]
[image]

MoonshineSG, 2018-05-12:
The print looks good. 
(German football fan?) :)

upD8R, 2018-05-12:
Haha, yes! Just preparations for the upcoming world championship next month.

IceflareTech, 2018-05-12:
The problem is extrusion. If I extrude 100mm in order to prime then E will be 100mm over its last position. When the first line in layer 2 runs it will retract 100mm, less the increment from the last line in layer 1, to get to the E value thus unloading the filament. I suppose I could retract the exact amount that I use in code while the filament is out. As long as my retracts equal my extrudes I should be ok. My printer runs absolute btw 

Edit: Just ran the code and all is great! Thanks to upD8R for the beep! took out the extruder set and made sure all was equal.

M300 S500 P2000 ;BEEP
G91                       ;Set to absolute positioning
G1 Z2 F200 E-10            ;retract 10mm of feed stock move 2
G90                       ;Set to relative positioning
G1 Y-200                  ;Move bed back
M117 Change Filament...   ;Display message
G91                       ;Set to absolute positioning
G1 F200 E-40              ;retract 40mm of feed stock
M300 S500 P2000 ;BEEP
M0                        ;Wait for buton press( is a resume in octoprint)
M117 Extruding...         ;Display message
G1 F200 E45              ;extrude 45mm of feed stock
M117 Press to return to print..         ;Display message
M300 S500 P2000 ;BEEP
M0                      ;Wait for buton press( is a resume in octoprint)
G4 P3000                  ;Wait 3 secounds
G1 F200 E5                ;extrude 5mm of feed stock
G4 P300                   ;Wait 1/3 secound
G90                       ;Set to relative positioning

closed 2018-09-10 by MoonshineSG