MoonshineSG/OctoPrint-MultiColors #27: Injecting GCODE failed ['ascii' codec can't decode byte 0xc4 in position 1557595: ordinal not in range(128)]

psy0rz, 2018-06-13:
I'm getting this error when injecting gcode.

Using OctoPrint 1.3.8 running on OctoPi 0.15.0PE

Multicolors 1.0.14

psy0rz, 2018-06-13:
Ah i just thought of something: I did a hexdump of the gcode and indeed discovered a 0xc4 character in it. 

I'm using Slic3r PE, and in the notes section of PLA there are some 'weird' ascii characters. (The notes will be added to the gcode as comments)

I removed the notes, tried again and now it works.

closed 2018-06-13 by psy0rz

romankysely, 2018-08-01:
Yes i had the same experience. The problem is that GCODE include characters in comments which describe for example name of vendors of filament. In my case  name of the vendor was "Mladeč". So and last character "č" is the problem . It is Czech republic national character which is out of 128 ASCII.
So I recommend to @MoonshineSG correct this in Multi Colors plugin or notice it. Most of  languages has these special characters used normaly so there will be a lot of troubles with it.  
Aaahhh we pay for our originality all the time  :-)) Thanks @psy0rz for key idea to resolve it.