MoonshineSG/OctoPrint-MultiColors #28: Multicolor gcode..;

Tripax46, 2018-09-09:
Hi
i have marlin 1.1.8
and M600 is activate
I have install Multicolors: https://plugins.octoprint.org/plugins/multi_colors/
But in advanced option i 've write : Layer
and
Add Gcode 
i 've write 
M117 Change Filament
M600
M0
but when i program to actiavte changement to layer 26 it s not run..My gcode is correct?
Thanks for your answer

MoonshineSG, 2018-09-10:
probably not... just "Layer" is not enought... it also depends on the code your slicer inserts. send me a sample and i'll try to get the right codes for you

Tripax46, 2018-09-10:
[image]
I have this config

Tripax46, 2018-09-10:
Ohhhh it's ik he run good!!! but i change Add gcode 
[image]