MoonshineSG/OctoPrint-MultiColors #29: No layer reference in Slic3r gcode

chadsittig, 2018-09-11:
I've output a gcode with Slic3r and noticed a difference in my gcode compared to the other examples I've seen.

My gcode output for between layers is as follows:

> ;BEFORE_LAYER_CHANGE
G92 E0.0
;1.2

G1 F8640
G1 X231.375 Y111.005 E-0.03360
G1 X231.375 Y109.366 E-0.37842
G1 X231.596 Y109.145 E-0.07214
G1 X231.667 Y107.953 E-0.27584
G1 E-0.04000 F2100.00000
G1 Z1.600 F10800.000
;AFTER_LAYER_CHANGE
;1.2

This refers to the layer height, rather than layer number which is what is being searched for in the regular expression. 

I've tried the following:
[image]

I get an error because I'm searching for a non-integer (e.g. 0.4 2.2).

MoonshineSG, 2018-09-11:
Slic3r allows custom gcode that you can add at layer change. They use [layer_num] as variable so you can customize the output.

bloodygoodbloke, 2019-11-08:
In the Printer settings Custom Gcode i have the following to display the layer changes on Octoprint Dashboard.  I added the extra line
:NEW_LAYER ;notifier_for_Octoprint_Multi_Color_plugin
 Then used that for the regular expression.

[image]

I have a dual extruder single head machine and have a purge bucket. the code i had in Slic3r i copied into the Inject section and made sure i Specified the Extruder As  "T1"