MoonshineSG/OctoPrint-MultiColors #31: ERROR: Injecting GCODE failed. Replaced 0 out of 1 needed.

g33k5h0p, 2018-10-21:
im using anet a8 stock firmware, blender then cura then octoprint with multicolors

gcode sample:

;LAYER:11
; multi color

M117 Change filament
M0
G0 X39.864 Y50.321 Z1.4

thanks in advance

MoonshineSG, 2018-10-22:
Try `;LAYER:{layer}` (no `space` between `:` and `{`)

MoonshineSG, 2018-10-22:
where did you this code ?
```
;LAYER:11
; multi color

M117 Change filament
M0
G0 X39.864 Y50.321 Z1.4```

g33k5h0p, 2018-10-22:
i just realized that my new laptop is running win7 64 bit my old one was 32 bit ,, would that make a diff?

MoonshineSG, 2018-10-22:
no, don't think the 64/32 would make a dif

send me a sample of your gcode file (before multicolor), a part with layer change so i can give you regex that will work...

g33k5h0p, 2018-10-22:
G0 X20.686 Y28.567
;TIME_ELAPSED:20601.707901
;LAYER:13
G0 X20.686 Y28.567 Z1.6

MoonshineSG, 2018-10-22:
no, don't think the 64/32 would make a dif

send me a sample of your gcode file (before multicolor), a part with layer change so i can give you regex that will work... 

(hope you get this mail - seams like github is down and i can't reply properly.)

g33k5h0p, 2018-10-22:
;LAYER:2
G0 F7200 X33.732 Y71.766 Z0.5
;TYPE:WALL-INNER
G1 F1500 E3351.77259
G1 F3600 X31.916 Y71.807 E3351.8028
G1 X30.444 Y71.763 E3351.82729

MoonshineSG, 2018-10-23:
try to slice a small sample object and send me the whole  .gcode file (zip it) via a dropbox or similar

g33k5h0p, 2018-11-06:
i have resurrected the old hdd ,, the only diff ikn setup is 32 or 64 bit ,, ill posts soon to see if it was just a platform issue

MoonshineSG, 2018-11-06:
Just send me goose sample and I’ll have a look

g33k5h0p, 2018-11-19:
sorry for the late response , work work work lol , 
this sample is from the last file i got to work 
0 F7200 X101.934 Y97.6
G0 X99.862 Y98.209
G0 X96.337 Y94.56
G0 X95.017 Y93.731
G0 X93.547 Y93.22
G0 X92 Y93.042
G0 X90.453 Y93.22
G0 X88.982 Y93.731
G0 X87.664 Y94.561
G0 X86.562 Y95.663
G0 X85.89 Y96.73
G0 X85.453 Y97.6
G0 X85.633 Y97.663
;TIME_ELAPSED:206.024650
;LAYER:4
G0 X85.633 Y97.663 Z0.7
;TYPE:WALL-INNER
G1 F1500 E210.59055
G1 F3600 X85.53 Y97.962 E210.59581
G1 X85.306 Y97.738 E210.59885
G1 X84.473 Y96.906 E210.61843
G1 X84.633 Y96.452 E210.62643
G1 X85.606 Y94.902 E210.65687

g33k5h0p, 2018-11-19:
ǝuıɥsuooɯ <notifications@github.com> wrote ..

> i requested a "goose sample"... not sure what that is... :D (note to self: don't
> reply support questions after a night out)
> 
> now back to your problem. send a sample file and not a copy/paste... spaces, special
> characters need to preserved as they are important... 
> 
> 
> 
> https://github.com/MoonshineSG/OctoPrint-MultiColors/issues/31#issuecomment-439753415
done

MoonshineSG, 2018-11-19:
strange.. i uploaded the gocde file you sent on my printer and `;LAYER:{layer}` works ... I'm running on a RaspberryPi... but the OS shouldnt make any difference... 

Can you see if there's any error message in the logs when it fails ?

g33k5h0p, 2018-11-19:
ǝuıɥsuooɯ <notifications@github.com> wrote ..

> i requested a "goose sample"... not sure what that is... :D (note to self: don't
> reply support questions after a night out)
>
> now back to your problem. send a sample file and not a copy/paste... [...]
having issues zipping for some reason ,,, i have a .txt for ya tho of an original file 

;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:3.4.0
;GENERATOR.BUILD_DATE:2018-06-26
;TARGET_MACHINE.NAME:Ultimaker 3
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:205
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:5795
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:AA 0.4
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:60
;PRINT.TIME:3783
;PRINT.SIZE.MIN.X:9
;PRINT.SIZE.MIN.Y:6
;PRINT.SIZE.MIN.Z:0.27
;PRINT.SIZE.MAX.X:139.627
;PRINT.SIZE.MAX.Y:144.824
;PRINT.SIZE.MAX.Z:3.87
;END_OF_HEADER
;Generated with Cura_SteamEngine 3.4.0
T0
M82 ;absolute extrusion mode

G92 E0
M109 S205
G0 F15000 X9 Y6 Z2
G280
G1 F1500 E-6.5
;LAYER_COUNT:26
;LAYER:0
M107
M204 S625
M205 X6 Y6
G0 F3750 X130.393 Y125.195 Z0.27
M204 S500
M205 X5 Y5
;TYPE:WALL-INNER
G1 F1500 E0
G1 F1200 X126.027 Y128.778 E0.22824
G1 X121.038 Y131.446 E0.45687
G1 X115.623 Y133.088 E0.68554
G1 X113.996 Y133.247 E0.7516
G1 X113.843 Y133.622 E0.76796
G1 X113.751 Y133.851 E0.77794
G1 X113.562 Y133.692 E0.78387
G1 X113.378 Y133.537 E0.79359
G1 X113.01 Y133.344 E0.81039
G1 X110 Y133.641 E0.93261
G1 X106.897 Y133.335 E1.05862
G1 X106.424 Y133.712 E1.08306
G1 X106.24 Y133.859 E1.09258
G1 X106.149 Y133.642 E1.09829
G1 X105.98 Y133.243 E1.1158
G1 X104.377 Y133.088 E1.18089
G1 X98.962 Y131.446 E1.40955
G1 X93.973 Y128.778 E1.63818
G1 X89.607 Y125.195 E1.86642
G1 X86.025 Y120.83 E2.09461
G1 X83.357 Y115.841 E2.32324
G1 X81.715 Y110.426 E2.5519
[... 2536 lines cut ...]
G0 F9375 X97.399 Y130.288
M204 S500
M205 X5 Y5
G1 F1500 X135.485 Y92.202 E384.11999
M204 S2812
M205 X18 Y18
G0 F9375 X135.313 Y91.88
M204 S500
M205 X5 Y5
G1 F1500 X97.077 Y130.116 E385.30026
M204 S2812
M205 X18 Y18
G0 F9375 X96.754 Y129.943
M204 S500
M205 X5 Y5
G1 F1500 X135.141 Y91.556 E386.48519
M204 S2812
M205 X18 Y18
G0 F9375 X134.969 Y91.234
M204 S500
M205 X5 Y5
G1 F1500 X96.432 Y129.771 E387.67475
M204 S2812
M205 X18 Y18
G0 F9375 X96.109 Y129.59

g33k5h0p, 2018-11-19:
sorry about that lol ,,,,, i tried what you have and i now get permission denied
would another plugin maybe cause this as a conflict issue?

g33k5h0p, 2018-11-19:
[image]

g33k5h0p, 2018-11-19:
[image]

MoonshineSG, 2018-11-19:
I see many atempts with dif "find" strings... just use `;LAYER:{layer}` Thats the one working for your file..

g33k5h0p, 2018-11-19:
permission denied still

MoonshineSG, 2018-11-19:
that i am not sure why...  maybe other plugins ?

g33k5h0p, 2018-11-19:
i can let you in on team viewer if you think that will help

MoonshineSG, 2018-11-19:
try without the "duplicate file" ...

g33k5h0p, 2018-11-19:
ya im using win 10 ,,, and 7 32 bit on the printer ,,, ima see if my other win 7 will do it

g33k5h0p, 2018-11-19:
nope still permission denied ,, ill reinstall from scratch tomorrow and see what happens

g33k5h0p, 2018-11-19:
i copied over my plugins and uploads folders from my previous installation, im a comp tech, and in my work doing recoveries and migrations ,,, sometimes i have to run a script called "take ownership" on everything imported,,, will explore that tomorrow and get back to ya ,,, thanks for all the help so far

closed 2018-12-03 by MoonshineSG

moisesbl, 2020-05-01:
Buenas. Estoy en la misma situación. ¿En cura donde tengo que modificar los valores? ¿Es incompatible con octolapse?

Jelenbl, 2021-10-14:
Try this as your regex statement.

.([LAYER])\w+.*([layer]).