MoonshineSG/OctoPrint-MultiColors #32: Is KiSSlicer supported?

bumbalu, 2018-12-02:
Hi!
I wonder if KiSSlicer is supported by MultiColors plugin. Looks like KiSSlicer doesn't number the layers. 
Below is the example of Gcode KiSSlicer produces at layer change. I need to insert @pause before the start of the layer z=2.2 Which regular expression should I insert into the "look for" field? I wasn't able to enter layer heights because the layer number in "look for" field is an integer. Any ideas/ help/ plans?
Thank you!

===============================================
; END_LAYER_OBJECT z=2.000
; Reset extruder pos
G92 E0
; BEGIN_LAYER_OBJECT z=2.200 z_thickness=0.200
; *** Estimate 4.318418 [s], Slowing (0.603x) to 7.0 [s] ***
; Perimeter Length = 84.589874
; Fan set to Cool value 100%
;
; 'Perimeter Path', 1.0 [feed mm/s], 24.1 [head mm/s]
G1 X109.972 Y75.028 Z2.2 E0 F4200
; head speed 24.120003, filament speed 1.002795, preload 0.131134
G1 X110.015 Y75.073 E0.04379 F320.1
====================================

bumbalu, 2018-12-02:
Sorry, I did not want to make the font that big. Can't change it for smaller.

MoonshineSG, 2018-12-03:
update to new version and use `; BEGIN_LAYER_OBJECT z={layer}.*`

bumbalu, 2018-12-03:
Alright, it works fine. Thank you! 
Just to mention for other KiSSlicer users that in the field "layer number or height" not the number but the layer height (ex. 2.2, 4.4) should be given.

closed 2018-12-03 by bumbalu