MoonshineSG/OctoPrint-MultiColors #33: need help getting multicolors to work ,,, fresh install of octoprint and cura

g33k5h0p, 2018-12-29:


MoonshineSG, 2018-12-30:
Some details could be useful

g33k5h0p, 2018-12-30:
i am running cura 3.1, and octoprint latest version, im wondering if the latest update to octoprint is the issue , as it worked great for me up until then ?

g33k5h0p, 2018-12-30:
ERROR: Injecting GCODE failed. Replaced 0 out of 1 needed.

[example.txt](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/2717332/example.txt)

MoonshineSG, 2018-12-30:
The regular expression used?

g33k5h0p, 2018-12-31:
[image]

merlin2105, 2019-03-13:
hi, same error her ..

ERROR: Injecting GCODE failed. Replaced 0 out of 1 needed.

MoonshineSG, 2019-03-14:
thats a problem with the regular expression matching the line from the slicer. you need to make sure the expression is right for your slicer

clarkkent435, 2021-01-06:
For Cura, use

;LAYER:{layer}

instead.

Jelenbl, 2021-10-14:
Try this as your regex statement.

.([LAYER])\w+.*([layer]).