MoonshineSG/OctoPrint-MultiColors #34: Using S3d to slice, need an example

thorkon, 2019-01-29:
I'm just getting started with using commands and gcode altering so this is still new to me but I want to start getting into color changing. I was wondering if I could get an example of what I would input into advanced options Find Line (regular expression) if I were using simplify3d and trying to pause at layer 14. Is this it: 
; layer {layer},.*

Then I'm assuming I need to input m commands in the inject gcode. Are the base ones that are shown (M117 Change filament
M0) enough to get a basic color change?

I'm reading up on the commands now and see that m0 stops the printer and m117 displays the message. I'm guessing I would probably want to input commands that would move the hotend away from the print.

I'm using a anycubic Chiron. (Can't recall the exact Marlin its running at the moment until I get to it at home, buts its whatever comes stock). Thanks for any help!

MoonshineSG, 2019-01-29:
Most of the times (?) M0 would do the “move head out of way” but it’s most of the time a “try and learn” game... it’s a matter of if you enjoy playing the game ;)

MoonshineSG, 2019-01-29:
And yes, the regents should work

MoonshineSG, 2019-01-29:
regents/regex/. (Thanks iOS spellchecker!)

thorkon, 2019-01-29:
Thanks for the fast resposne! I'll give this a shot and see how it goes :D.

thorkon, 2019-01-29:
Am I able to pause the layer height at different heights with this? (10,23,37,etc)?

thorkon, 2019-02-04:
https://imgur.com/xSndJ5Y

I get an error whenever I hit process. Anything look wrong?

MoonshineSG, 2019-02-11:
need more info... try to look at the logs and post here any error that has been logged

komandrik, 2020-06-01:
I get the same error as @thorkon
Below is a piece of log. How can I fix it. What am I doing wrong?

2020-06-01 19:22:09,672 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|      cores: 4
|      freq: 1400.0
|      ram: 917008384
|  os:
|      id: linux
|      platform: linux2
|  plugins:
|      pi_support:
|          model: Raspberry Pi 3 Model B Plus Rev 1.3
|          octopi_version: 0.17.0
|  python:
|      pip: 19.3.1
|      version: 2.7.16
|      virtualenv: /home/pi/oprint
2020-06-01 19:32:36,618 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'process' / '{u'layers': u'15,18', u'gcode': u'M117 Change filament\nPAUSE', u'duplicate': True, u'command': u'process', u'file': u'TestCubeXYZ2cm2.gcode', u'find_string': u';layer {layer}.*?'}'
2020-06-01 19:32:36,629 - octoprint.plugins.multi_colors - INFO - File to modify '/home/pi/.octoprint/uploads/TestCubeXYZ2cm2.gcode.tmp'
2020-06-01 19:32:36,704 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '15'...
2020-06-01 19:32:36,755 - octoprint.plugins.multi_colors - INFO - Failed to insert code for layer 15
2020-06-01 19:32:36,759 - octoprint.plugins.multi_colors - INFO - Trying to insert multi color code for layer '18'...
2020-06-01 19:32:36,809 - octoprint.plugins.multi_colors - INFO - Failed to insert code for layer 18

komandrik, 2020-06-01:
I accidentally found a solution. I looked at the G-code and found that there was still a space before the layer.
```
G1 E-0.9000 F2400
; layer 3, Z = 0.600
G4 P1
```
Added a space before the layer to the expression and the plugin worked.
`; layer {layer}.*?`