MoonshineSG/OctoPrint-MultiColors #35: What Slicer was this made to work with?

ldboehm, 2019-02-27:
What Slicer did you develop this around?

MoonshineSG, 2019-02-27:
i am using Simplify3D, but because it's configurable (using regex) it should work with any slicer

clarkkent435, 2021-01-06:
For Cura, use

;LAYER:{layer}
instead of the Simplify3d default.

jstuewe, 2021-02-01:
I'm using PrusaSlicer 2.3.0 on my Ender 3 R2 and had to play around with the REGEX expression.
This slicer puts the layer comment on the line with the layer change Gcode.
ie. "G1 Z1.000 F9000.000 ; move to next layer (4)"

To get this slicer to work the correct REGEX expression is:

move.to.next.layer.\({layer}\)

[image]