MoonshineSG/OctoPrint-MultiColors #36: Injecting GCODE failed

turuncuoglu, 2019-05-03:
I am using an Anet a8 with merlin and s3d for slicer  My g code is 
G1 X91.892 Y91.740 E18.9511
G1 X91.855 Y91.740 F2880
G1 X91.740 Y91.855 F2880
G92 E0
G1 E-10.0000 F3600
; layer 5, Z = 0.784
; inner perimeter
G1 X91.400 Y91.400 F4800
G1 Z0.784 F1002
G1 E0.4000 F3600
G92 E0
G1 X108.600 Y91.400 E0.4559 F2700
G1 X108.600 Y108.600 E0.911

[image]

I can not insert the layer.  

Thanks for your help.

MoonshineSG, 2019-05-04:
`; layer {layer},.*?` should work... make sure the number of spaces is the same in the code and the regex

MoonshineSG, 2019-05-04:
if that still doesnt work, send me a full gcode file so i can do some tests...

turuncuoglu, 2019-05-04:
Thanks for the quick response.  I was able to inject the code with modifications.  The printer pauses but it is at the beginning of the layer.  Is there a way to pause the printer move it out of the way and return when the print is resumed?

[xyzCalibration_cube_multi.zip](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/3144428/xyzCalibration_cube_multi.zip)

MoonshineSG, 2019-05-04:
The injected code is customisable as well... :)

turuncuoglu, 2019-05-04:
Yes right after I wrote that I realize that the code is customizable.  I will see if can figure it out.  Thanks for your help and great plug in.

closed 2019-07-09 by MoonshineSG