MoonshineSG/OctoPrint-MultiColors #38: Not injecting g-code Issue.

mstein792, 2019-06-17:
I am having the issue of it failing to inject. I am using S3D and the expression used in my gcode is "; layer 13, Z = 2.660" and in the plugin I am putting "; layer {layer}" I have tried most fixes i've found and no dice...  any ideas?

MoonshineSG, 2019-06-18:
send a sample of your file and i'll have a look

MoonshineSG, 2019-06-18:
Btw, I'm using S3D and my - working - regex is `;LAYER:{layer}`

mstein792, 2019-07-04:
[30 doneee.zip](https://github.com/MoonshineSG/OctoPrint-MultiColors/files/3357722/30.doneee.zip)
here ya go, thanks!!

mstein792, 2019-07-04:
also, tried the one you posted and no dice.  Just to make sure i'm doing this right... I upload to octoprint, choose the file, then go to multicolor and just input layer number and correct regex, right?!?

MoonshineSG, 2019-07-08:
try `; layer {layer}.*`

mstein792, 2019-07-08:
Worked perfect!! Thank you!

closed 2019-07-09 by MoonshineSG