MoonshineSG/OctoPrint-MultiColors #39: Printer pauses at 1, 10, 11 ,12, etc.

TheNewKobra, 2019-07-31:
Hey,
first of all thanks for the great plugin!
Today I found out that if I inject a pause code at layer number 1 it will also inject it at every layer number starting with a 1, eg. 11, 12, 13, etc. stopping at layer number 20.

I guess the code is just looking at the first digit and stops reading thereafter...

MoonshineSG, 2019-07-31:
Show me your regex and a sample of your file

drjrkuhn, 2020-12-02:
Great idea for a plugin! But I am seeing the same problem as @TheNewKobra 

For Simplify3D, the layers are marked as
```
; layer 1, Z = 0.240
; layer 2, Z = 0.480
...
; layer 10, Z = 2.400
; layer 11, Z = 2.640
; layer 12, Z = 2.880
```

I am using the regex string `; layer {layer}.*?`. The problem is that I need to include a comma `,` at the end of the regex pattern to end the match, but that doesn't work. The plugin doesn't find any layer text if I add a comma after with `; layer {layer}.*?,`. I am not quite sure how `{layer}` translates to regex pattern substitution.

MoonshineSG, 2020-12-09:
yours shodul workd with `; layer {layer},.*` .{layer} gets replaced with the layer number you enter, before the regex is used. So if you look for layer 5, the matching regex will be `; layer 5,.*`

drjrkuhn, 2020-12-23:
Thanks for the explanation. That worked!