MoonshineSG/OctoPrint-MultiColors #41: REQUEST:  Multi changes using multi extruders

bloodygoodbloke, 2019-11-08:
Can it be included in a future version the ability to change multiple times in a single print but being able to Choose the extruder to use?

MoonshineSG, 2019-11-09:
the plug-in is designed to allow multi colour prints using a single extruder... 

Can you give a use case for a job on a dual extruder printer where you would need to use this plugin? Then I’ll see how that could be implemented.

bloodygoodbloke, 2019-11-11:
I have a multi extruder single head machine and using this plugin i can turn a single colour print into a multi colour especially when its a layer change without reslicing it.     
the other use is using this plugin to make a Tri or even Quad colour.  when the code changes to a new extruder i can change the filament on the first extruder and get it to flip back.  
up to now i have used Prusa-color page to do these things but this plugin is so much handier

MoonshineSG, 2019-11-12:
when the printer is paused (by the plugin via the inserted gcode) you can manually issue a T1 or T2 command to manually change the extruder...